PaulStoffregen/SPI #5: SPIClass does not name a type

dderiso, 2015-03-11:
When compiling with a Teensy 3.1, I received the following error: "SPIClass does not name a type". Reverting back to your version on Dec 21, 2014 solves the problem. Thanks for all the hard work!

PaulStoffregen, 2015-03-11:
Please post the code you're compiling

PaulStoffregen, 2015-03-11:
also, which teensy you're using and which version of teensyduino matters,...

PaulStoffregen, 2015-03-11:
1.21-beta10 was released today.  please confirm the problem happens with -beta10.  If so, post details and I will investigate and fix whatever is wrong.

dderiso, 2015-03-11:
Hi Paul, Thanks for the quick response. The error was invariant to the code I was compiling with since it's an issue with the header itself (ie. it will throw an error even if I have an empty sketch). I'll upgrade next week and will let you know if the problem persists, otherwise if I forget to update you, please assume it's all working with the new version :) Cheers, Dave

PaulStoffregen, 2015-03-14:
I just compiled File > Examples > SPI > DigitalPotControl for Teensy 3.1 on Arduino 1.6.1 with TD 1.21-beta12.  No errors.

I'm going to close this issue.  If you find a case that doesn't work with the latest version, please open a new issue with specific details to reproduce the problem.

closed 2015-03-14 by PaulStoffregen