PaulStoffregen/SPI #7: Suggestion: Add 8- or 16-bit transfer size setter

crteensy, 2015-03-22:
Can we have a method that sets the SPI to 8-bit or 16-bit mode, possibly with an enum argument that selects between these transfer sizes? These sizes are the common denominator for Teensy 3.x and Teensy LC.

Transfer size can probably not be added to SPISettings, because that's a part of the common arduino interface, isn't it?

TheSeventhCG, 2019-01-25:
Perhaps it could be a nice idea to implement a real **16Bit SPI transfer function** into the SPI library. Not like the current one, which is splitting the 16Bit word into a high and low 8Bit variables to send them.

Actually modifying the library to transfer 16Bit is not the problem, but if possible it would be nice to have a possibility to send **8Bits** with **SPI.transfer** and real **16Bit** with **SPI.transfer16**.

This is only an idea from my side in addition to what @crteensy said.

PaulStoffregen, 2019-01-25:
I don't understand why you're asking this?  On the hardware which supports 16 bit transfers, we do use it.

PaulStoffregen, 2019-01-25:
You're asking in the wrong place regarding Arduino Due.  This is the SPI library for Teensy boards.

PaulStoffregen, 2019-01-25:
And yes, I would agree, the library for Due leaves much to be desired.  But I'm not the one to fix it, and this is not the place to request changes to a completely different copy of the SPI library.

TheSeventhCG, 2019-01-25:
Yes, you are right, this is the wrong place 

sorry about that

PaulStoffregen, 2019-01-25:
Maybe consider giving Teensy 3.5 or 3.6 a try?  Both are more powerful than Due, and they use this SPI library which does support real 16 bit transfer.  It also has many other performance improvements.  I'm sure you'll find SPI on Teensy works much, much better!  :-)