PaulStoffregen/SPI #32: SPI1 & SPI2 need compiler better compiler error messages.

Donziboy2, 2017-12-08:
If you use SPI1 or SPI2 and compile for a Teensy 3.2/3.1 it gives a generic error .

error: 'SPI1' was not declared in this scope
 SPI1.begin();

If possible it would be nice to say these functions are only usable in T3.5/T3.6 instead of a generic error message that we get.  Took me about 2 hours to figure out why it was throwing errors.

tonton81, 2018-04-12:
that hardware is defined for devices that use it

if your attempting to access hardware not available, obviously youd get that error.
This is not a “bug”, not should code be bloated to tell users to switch to a different MCU tonuse hardware that doesnt exist on theirs...

PaulStoffregen, 2018-04-17:
There is some code in the Arduino IDE to detect certain cases and print friendly messages.  It can be found in message_advice_Teensy() in Compiler.java.

https://github.com/PaulStoffregen/Arduino-1.8.5-Teensyduino/blob/master/arduino-core/src/cc/arduino/Compiler.java#L686

But as you can see, it's not written to be customized depending on which board is selected.  It also looks only for fixed strings in English, which doesn't work when the compiler it localized to another language.  Eventually I would like to greatly improve this code, allowing the IDE to parse the compiler messages in any language and to get these messages from a text file that can be customized per platform and per board without editing the Java code.

Sadly, this is a low priority compared to so many other projects.....