PaulStoffregen/SPI #56: SSD 1306 Oled not working properly with HW SPI and Teensy 4.0

Guzu1987, 2020-02-11:
I have an 2.42 inch Oled display wired to the Hardware SPI port of the Teensy 4.0. 
SCK   13
MOSI 11
CS      10
DC     8
RST    9

When I use the Adafruit SSD1306 only the top 8 pixels of the oled work. The rest is noise. 
With the u8g2 library horizontal lines appear on the oled no matter what I do.
The lines are every 8th pixel (7 pixels between the lines).

The oled works fine with an Arduino Nano with both libraries. 

[image]

The only library which seems to work fine is the Sparkfun HyperDisplay library. But this one is VERY slow. 

After much trial and error (which can be followed [here](https://github.com/olikraus/u8g2/issues/1104)) I think this is a hardware SPI Issue with the Teensy code. (If I use the Software SPI functions _with the same or different pins_ in the u8g2 library everything works fine, just slower.)

I tired to upload the code with lower Teensy microprocessor speed (down to 24MHz chosen from the Arduino IDE) but this had no effect on the horizontal lines.

[This ](https://drive.google.com/file/d/13Im5G-Qq5yMSgdjmagW6BAxApm9EsB6n/view) and [this ](https://www.displaymodule.com/products/2-42-dm-oled24-630)is the oled I use and [this ](https://drive.google.com/file/d/1asGx-ihNh5kXNCYV3bchUqQuPBEiaYy6/view )is some documentation regarding the driver. 

Could there be an issue with the SPI software written for the Teensy 4.0? 

I would really appreciate any help in solving this issue. Thank you very much.

PaulStoffregen, 2020-02-11:
Duplicates this forum thread:
https://forum.pjrc.com/threads/59357-Problem-using-Teensy-4-0-with-2-42-quot-Oled-and-u8g2-or-Adafruit-Library

Guzu1987, 2020-02-11:
I solved the problem with help from the PJRC forums. 
I changed the spi_mode to 3 in the u8g2 library cpp files. It turns out that this particular oled uses this spi_mode. 
So the problem was not with the SPI library, for the Teensy but with my knowledge of the oled and code :)
[Here](https://forum.pjrc.com/threads/59357-Problem-using-Teensy-4-0-with-2-42-quot-Oled-and-u8g2-or-Adafruit-Library?p=229569#post229569) is the thread that helped me. 

Thank you very much.

olikraus, 2020-02-11:
I personally think that modes 0 and 3 are swapped. There was a similar issue with the ESP8266 boards (https://github.com/esp8266/Arduino/issues/2416). 

The SPI modes had been defined long back for the Arduino AVR boards and all other boards should behave in the same way.

The byte 0x053 is written in all four SPI modes: From left to right mode 0, 1, 2 and 3 on an Arduino Uno:
[image]

It would probably simple to crosscheck this with Teensy 4.0.

I personally would keep this open until it is verified, that the SPI modes behave in the same way.

PaulStoffregen, 2020-02-12:
I ran a quick test with this code:

```
#include <SPI.h>

void setup() {
  pinMode(10, OUTPUT);
  SPI.begin();
}

void loop() {
  SPI.beginTransaction(SPISettings(4000000, MSBFIRST, SPI_MODE3));
  digitalWrite(10, LOW);
  SPI.transfer(0x5A);
  digitalWrite(10, HIGH);
  SPI.endTransaction();
  delay(1);
}
```

On Arduino:
[image]

On Teensy 4.0:
[image]

On Teensy 3.2:
[image]

Hopefully it's obvious the state of MOSI outside of the 8 clocks does not matter.  In any case, Teensy 4.0 and Arduino Uno both keep the MOSI signal low, so exactly the same waveforms (other than digitWrite is so slow on Uno).

The waveforms are correct.  I'm considering this issue closed & fully resolved.

closed 2020-02-12 by PaulStoffregen

olikraus, 2020-02-12:
@PaulStoffregen thanks for the quick response. Mode 3 indeed looks correct. However it actually is Mode 0 which had caused the problem. I wonder how the Teensy 4.0 picture would look like for mode 0.

BTW: Amazing little board. Good Work.

PaulStoffregen, 2020-02-12:
Please run the test program and capture the waveforms.  If there is a problem with the waveforms, show me that *you* have done the test.

I have many other projects needing my time, and many other people to help on the forum, so I'm considering this matter resolved.  I'm not going to put any more dev time into it, unless you can show waveforms which demonstrate a problem.

olikraus, 2020-02-12:
Understand, same time problem here.
Moreover: I do not own a Teensy 4.0.
@Guzu1987 : Are you able to capture the waveform?

Guzu1987, 2020-02-12:
I am sorry but I don't have an oscilloscope. I only have one of these:
It has only 1 channel. If you think that I could measure something that would help you, just let me know. I think here are the specs of the DSO150 I have.
https://jyetech.com/dso-150-shell-oscilloscope/

[image]

olikraus, 2020-02-12:
Nice device, but we need to check clk vs data, so two channels are required.

Ok, let's consider this as solved...